argonne-vci/node-red-contrib-modbustcp #5: Auto Reconnect

thegunshow, 2016-05-23:
Hi

I am reading data using the Modbus "master" node. if it losses the connection will it automatically try and reconnect, do you know what the reconnect time is?

Thanks
Great node to have!!

thehellodonkey, 2016-08-05:
Agreed that this is a great node to have. Unfortunately, for me at least, if the connection is lost it will never reconnect without restarting/redeploying the node manually. It would be extremely helpful in my use case (and probably most others) if there were auto-reconnect functionality.
Thanks!

jayharper, 2016-12-13:
Added auto-reconnect functionality with recent update.  You can also set the Reconnect Interval (s) on the connection config tab.

closed 2016-12-13 by jayharper