argonne-vci/node-red-contrib-modbustcp #13: Holding register and bits

tomasinouk, 2017-07-16:
Hi, is there a way to read specific bits in holding register?
Thank you

marcusjelsch, 2017-07-16:
Yes. U can try to read it like a coil.

tomasinouk, 2017-07-16:
hi, thanks for quick response,
I tried that and no luck there.

for example the holding register address is 256
e.g. 40256
I tried to read the coil 256 with quantity of 15 (assuming reading bits from 0-14)

I got nothing back though, any suggestion?
Thank you

marcusjelsch, 2017-07-16:
Does ur modbus device support coils?

tomasinouk, 2017-07-16:
it doesn't only holding registers

marcusjelsch, 2017-07-16:
Search my threads. I have had the same problems. There u will find the
solution.

tomasinouk, 2017-07-16:
will do and post here, if I find a solution

jayharper, 2017-07-17:
Why don't you just read the entire holding register and use a function block with a bit mask to read the specific bit you are interested in?

marcusjelsch, 2017-07-18:
Do u have an example?

tomasinouk, 2017-08-01:
Hi, sorry for long delay on my side.

@jayharper thank you for suggestion. That led me to using function node with this code

```javascript
var a = msg.payload[0];
var b = [];
for (var i = 0; i < 16; i++)
  b[i] = (a >> i) & 1;

var a = msg.payload[1];
var c = [];
for (var i = 0; i < 16; i++)
  c[i] = (a >> i) & 1;

b = b.concat(c);
msg.payload = b
return msg;
```
I am reading two registers, let say 500 and 501, that is the array in `msg.payload[]` and each register has 16bits, which I am accessing in `for loop`.

I do have sneaky suspicion, that I am not applying your suggestion correctly though. 

I can post whole flow, if needed.

jayharper, 2017-08-03:
Try this.  Feel free to edit as necessary.
`msg.payload = [];
msg.payload[0] = 0xAACC;
msg.payload[1] = 0X00FF;

function isNthBitSet (c, n) {
    var mask = [1,2,4,8,16,32,64,128,256,512,1024,2048,4096,8192,16384,32768];
    return ((c & mask[n]) !== 0);
}

var b = [];
for (var i = 0; i < 16; i++)
{
  b[i] = (isNthBitSet(msg.payload[0], i));
}

var c = [];
for (var i = 0; i < 16; i++)
  c[i] = (isNthBitSet(msg.payload[1], i));

b = b.concat(c);
msg.payload = b;
return msg;`

closed 2017-08-09 by jayharper