argonne-vci/node-red-contrib-modbustcp #27: Read AND Write Connection to one Modbus-Server

Markus-, 2019-05-11:
I have an AC which is controllable by ModBus.

So I tried the Read-Node and can query the AC without any problems.
When I implement a Write-Node it can not connect to the Server as long as the Read-Node works.

How could I achieve reading in intervals and writing as needed to the same server?
Is this possible?

Regards
Markus

bnystrom, 2019-07-22:
As long as you are using a single (the same) server connection, this should work. We've seen modbus servers that only allow a single connection. Curious if you created two separate server connections or if you put the read and the write nodes on separate tabs?

Markus-, 2019-07-24:
I realized it with Flex Getter / Flex Write - this way it's working.

closed 2019-07-24 by Markus-