argonne-vci/node-red-contrib-modbustcp #29: Modbus write API wont work

OceeL, 2019-07-19:
Hello!

Im am trying to use the Modbus write node and its API.

When I inject integer as a payload to the configured Modbus tcp write node, it will send the integer to proper register to the servers end so the communication with the server works. 

How ever when I try to send message using the API, so that the values in the message would over write the ones in the modbus write node it wont work. Is there something I am doing wrong? There is no error or anything else that would tell me whats wrong.
[image]
_Configuration of nodes_
[image]
_msg function node_
[image]
_Content of msg in the msg debug node_

Thank you!
Br,
Oskari

bnystrom, 2019-07-22:
Oskari,

I've just tested this and do not see any issues.

Here is my simple flow with a couple examples:

[{"id":"20c4a58a.ce4112","type":"tab","label":"Flow 3","disabled":false,"info":""},{"id":"4ad4e426.55a974","type":"modbustcp-read","z":"20c4a58a.ce4112","name":"Read One","topic":"","dataType":"HoldingRegister","adr":"1","quantity":"1","rate":"2","rateUnit":"s","server":"e8dde1b8.85d298","ieeeType":"off","ieeeBE":true,"x":160,"y":550,"wires":[["29148ba1.a59b4c"]]},{"id":"bbd617ac.81a15","type":"modbustcp-write","z":"20c4a58a.ce4112","name":"Write One","topic":"","dataType":"Coil","adr":"1","server":"e8dde1b8.85d298","x":540,"y":190,"wires":[]},{"id":"b7aedf7d.c487e8","type":"inject","z":"20c4a58a.ce4112","name":"","topic":"","payload":"1","payloadType":"num","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":150,"y":190,"wires":[["bbd617ac.81a15"]]},{"id":"29148ba1.a59b4c","type":"debug","z":"20c4a58a.ce4112","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload[0]","targetType":"msg","x":440,"y":550,"wires":[]},{"id":"a23baf2e.aadf","type":"inject","z":"20c4a58a.ce4112","name":"","topic":"","payload":"0","payloadType":"num","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":150,"y":230,"wires":[["bbd617ac.81a15"]]},{"id":"30866b7b.7e29e4","type":"function","z":"20c4a58a.ce4112","name":"From Payload","func":"var payload_in = msg.payload;\nmsg = {\n    address: 1,\n    dataType: \"HoldingRegister\",\n    payload: payload_in\n    \n};\n\n\nreturn msg;","outputs":1,"noerr":0,"x":340,"y":320,"wires":[["bbd617ac.81a15"]]},{"id":"50e9104f.806458","type":"inject","z":"20c4a58a.ce4112","name":"","topic":"","payload":"1","payloadType":"num","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":150,"y":310,"wires":[["30866b7b.7e29e4"]]},{"id":"559566c3.80a21","type":"inject","z":"20c4a58a.ce4112","name":"","topic":"","payload":"0","payloadType":"num","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":150,"y":350,"wires":[["30866b7b.7e29e4"]]},{"id":"edecd98.a8b00a8","type":"function","z":"20c4a58a.ce4112","name":"","func":"var msg = {\n    address: 1,\n    dataType: \"HoldingRegister\",\n    payload: 1\n}\nreturn msg;","outputs":1,"noerr":0,"x":360,"y":430,"wires":[["bbd617ac.81a15"]]},{"id":"561d181e.91f6b","type":"inject","z":"20c4a58a.ce4112","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":160,"y":430,"wires":[["edecd98.a8b00a8"]]},{"id":"edcf0355.4dab78","type":"comment","z":"20c4a58a.ce4112","name":"Basic no-API","info":"","x":170,"y":140,"wires":[]},{"id":"75e5282b.8eec68","type":"comment","z":"20c4a58a.ce4112","name":"API w/ variable input","info":"","x":190,"y":270,"wires":[]},{"id":"2be9943e.f5d29c","type":"comment","z":"20c4a58a.ce4112","name":"API w/ hardcoded payload of 1","info":"","x":230,"y":390,"wires":[]},{"id":"aaabb09d.61dcd","type":"comment","z":"20c4a58a.ce4112","name":"Read addr:1 -> Output","info":"","x":190,"y":500,"wires":[]},{"id":"e8dde1b8.85d298","type":"modbustcp-server","z":"","name":"ideapadLT","host":"130.202.169.70","port":"502","unit_id":"1","reconnecttimeout":""}]

OceeL, 2019-07-24:
Hello and thank you for your answer!
I have done some testing and it seems that my device vendor has installed old version of this node, and this is causing the problem. 

The code you provided works on a platform that has the newest version of this node, and wont work with the device that has the old version of this node.

I consider this issue solved now.

closed 2019-07-24 by OceeL

bnystrom, 2019-07-24:
Oceel,

Thanks for letting us know. Out of curiosity, what "device" / "vendor" is this being installed on?

OceeL, 2019-07-26:
The decive in question is advantech's ICR-3200 series IoT 4G Gateway.
[https://www.advantech.eu/products/industrial-iot-4g-lte-router-and-gateway/sub_7e21d2de-630c-4925-a51c-f41c18642d72](url)
It is no ordinary Node-Red platform and the nodes have to be "repackaged" by the manufacturer to work. It seems that the node they provided for this device is base on older version of your node and wont support the usage of API in modbus TCP Write node.