argonne-vci/node-red-contrib-modbustcp #30: {"err":"Offline","message":"no connection to modbus server"}

Sadafaga, 2019-08-09:
Hello,

 I am using the package "node-red-contrib-modbustcp" . I create a modbus tcpip server and it worked fine until it appeared the following error: "{"err":"Offline","message":"no connection to modbus server"}" and also "socket error: Error: connect ECONNREFUSED" . The server was configure with 127.0.0.1 and port 502. The firewall of the computer is turned off.

I reboot my computer and It works again, but I would like to know what it's the reason to avoid it in the future !

running on windows 10
 node version: 8.11.3
node-red version: 0.20.3 

Regards

Ekristoffe, 2020-03-29:
Hello
Late answer I know, but:
I had the same trouble and see the modbus system use a socket for each modbus read or write.
Because of this, the modbus server is easily overflow with socket used only for one thing.
I think having one socket per server would be nice.
I will see if I can debug this and test it with an system where I can limit the server socket to 1 (only 1 socked per server would be perfect)

gr-hayduk, 2022-09-12:
I have the same issue. I can use modbustcp client Read registers (all types), but when I try to inject a value into any Register Write node, then immediatelly I get this

[[error] [modbustcp-write:172dd4e19fe8c323] : {"err":"Offline","message":"no connection to modbus server"}]

and then also Read Register nodes log their "...was disconnected or was unable to connect", ... was Disconnected.

I have tested two different modbus slaves (unslave.xyz and commercial PLC controller). I haven't any problems with them until node-red - modbus. Sad.