argonne-vci/node-red-contrib-modbustcp #32: Modbus Server crashes because of too many connections after often deploy

nischram, 2019-09-30:
Hi,

while I created a new Flow with one or more "modbustcp client" Node and had to edit it many times my Modbus TCP server crashed.

I found out that after a change to the flow or other flows and the "deploy" a new connection is opened but the old one is not closed in advance.

If I do a lot of testing and are often deployed, there are sometimes so many connections open that problems can occur on the server.
So I have to restart the system after some work so the system or the Modbus server does not crash.

Here is some information: I use Node-Red on a Homematic with the modbustcp as client to the modbus server (E3DC-S10) all programs and systems are up to date.

Here is an abstract of the open Modbus connections after 2 changes and deploy, in the flow are 3 Modbus client with different queries:
```
# netstat -ant
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 192.168.178.22:46378    192.168.178.33:502      ESTABLISHED
tcp        0      0 192.168.178.22:46374    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:46376    192.168.178.33:502      ESTABLISHED 
# netstat -ant
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 192.168.178.22:46378    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:51620    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:46374    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:51622    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:51618    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:46376    192.168.178.33:502      ESTABLISHED 
# netstat -ant
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 192.168.178.22:46378    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:54144    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:51620    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:54142    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:46374    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:54140    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:51622    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:51618    192.168.178.33:502      ESTABLISHED 
tcp        0      0 192.168.178.22:46376    192.168.178.33:502      ESTABLISHED 
#
```

thanks
best regards
Nico

Ekristoffe, 2020-03-29:
Hello
Late answer I know, but:
I had the same trouble and see the modbus system use a socket for each modbus read or write.
Because of this, the modbus server is easily overflow with socket used only for one thing.
I think having one socket per server would be nice.
I will see if I can debug this and test it with an system where I can limit the server socket to 1 (only 1 socked per server would be perfect)
I think this one is related to this : #30

nischram, 2021-05-20:
Hey,
I have now tested it with the RedMatic version 7.2.1 and the Modbus-TCP version 1.2.3. 
When I deploy the flow or restart NodeRed, no new connection is opened and the Modbus client is no longer overloaded and runs without problems. Only 1 connection is open at any time.
thanks and best regards
Nico

closed 2021-05-20 by nischram