argonne-vci/node-red-contrib-modbustcp #34: Connection Crashes

slogan87, 2020-05-07:
Hi There,

I have just installed your node to read and write values out of a PLC. When ever I deploy changes to my code, I get the following

[error] [modbustcp-server:Pool] socket error: Error: read ECONNRESET node red

and the connection becomes intemittant. Stopping Node Red and restarting it fixes it. I was wondering if you could help me sort this issues

Regards
Sam

jayharper, 2020-05-08:
Hi Sam,

It’s very hard to determine how each Modbus device reacts to a socket being closed.  Have you tried deploying your changes by selecting the “Modified Flows” option.  Click the down arrow of the Deploy button to select your deployment option.

Jason Harper
Office: 630-252-1032<tel:630-252-1032>
Fax: 630-252-3443<tel:630-252-3443>

kskenyon, 2020-09-22:
I'm having the same problem.  Changing anything in settings and then deploying fixes it.  The ModbusTCP server is working fine.  This node is reading intermittently and only with some manual intervention.

Kevin